eworm-de/routeros-scripts #1: Updates for global-config

Azylog, 2019-03-28:
Hello,
Just discovered your Mikrotik script repo and it looks really nice !

Can you just elaborate on this :
:global ScriptUpdatesIgnore {
  "global-config"
}

Why is this file not updated on script-update ? How are we supposed to change the destination address for mail backup ?

Thanks a lot for your work

eworm-de, 2019-03-28:
You are supposed to update or modify the script manually, with `/ system script edit global-config source`. Having `global-config` in `$ScriptUpdatesIgnore` makes sure it is not overwritten by running `script-updates`.

Whenever something essential changes in `global-config` I will increase `$ExpectedConfigVersion` (and `$GlobalConfigVersion`). Then `script-updates` will notice and notify to have a look and merge required changes manually.

closed 2019-03-28 by eworm-de

Azylog, 2019-03-29:
So, as far as I understand, the global-config file is more a file containing parameters for a specific routeros.

Perhaps should we distinguish specific variables (like the DailyPskSecrets) and variables for a pack of devices (like ScriptUpdatesBaseUrl or EmailGeneralTo) ?

The local parameters file shouldn't be updated regulary but the update URL can change over time so having this setting being update on a regulary basis could be interesting.

I can do it on my side but perhaps are you more interested to include it in your script base.

eworm-de, 2019-03-29:
All variables in `global-config` are supposed to be changed by users. Let's have a look at `$ScriptUpdatesBaseUrl`: You could fork my repository to maintain your own changes, then modify the variable in `global-config` to point to your repository:
```
:global ScriptUpdatesBaseUrl "https://github.com/Azylog/routeros-scripts/raw/master/"
```
That's nothing you want to be overwritten, no?

If I do make incompatible changes in `global-config` I will inform uses by incrementing `$ExpectedConfigVersion`. Everybody will have to merge changes on its own.

Or do you have specific suggestions how to handle this better?

eworm-de, 2020-01-28:
With 1cee36a911ef20374cbafc989a5d3699c3fc0def you can use `global-config-overlay`. I guess that's what you were looking for.