eworm-de/routeros-scripts #2: Email Backup error

Kampfwurst, 2019-07-17:
Hello,

If i try to trun "/ system script run email-backup" i get the error "Error sending email [xxx] Backup & Config>:invalid BODY; I seems that there is a problem with the file.

Im using version 6.44.5 and 6.45.1. Sending a test mail over the menu is working fine.

Thanks a lot for your work

eworm-de, 2019-07-17:
I have never seen this myself. This must be device or configuration specific. Is there anything bogus with identity name, board and model name or serial number?

You could try to edit the script and comment the lines with extra information to find the causing value.

Kampfwurst, 2019-07-17:
Hi, The causing value is the file backup file.
If i remove the line " file=$Attach;" in tool e-mail then the script runs fine without backup file ;-(
The name config file name is "MikroTik.example.com.rsc" so this should be fine.

The board is a  RB433.
Hostname:      MikroTik
Channel:       long-term
RouterOS:      6.44.5

eworm-de, 2019-07-17:
You have the default configuration for backup?
```
:global BackupSendBinary false;
:global BackupSendExport true;
```

Does it help if you replace `file=$Attach` with `file=[ :tostr $Attach ]`?

Kampfwurst, 2019-07-17:
Yes I use the default configuration.

```
:global BackupSendBinary false;
:global BackupSendExport true;
```

`# send email with status and files
/ tool e-mail send to=$EmailBackupTo cc=$EmailBackupCc \
    subject=("[" . $Identity . "] Backup & Config") \
    body=("Backup and config export for " . $Identity . ".\n\n" . \
      "Board name:    " . $BoardName . "\n" . \
      "Model:         " . $Model . "\n" . \
      "Serial number: " . $SerialNumber . "\n" . \
      "Hostname:      " . $Identity . "\n" . \
      "Channel:       " . $Channel . "\n" . \
      "RouterOS:      " . $InstalledVersion . "\n\n" . \
      "Backup file:   " . $BackupFile . "\n" . \
      "Config file:   " . $ConfigFile) \
    file=[ :tostr $Attach ];
}`

still the same error.

eworm-de, 2019-07-17:
Does your device's identity contain bogus (possibly invisible) characters? Run...
```
:global Identity "test";
```
... then try again.

Kampfwurst, 2019-07-18:
still the same

eworm-de, 2019-07-18:
The device performance is not relevant, I am using this on RB751 with 400 MHz CPU...

Would you mind sending me a mail without backup and copy? Edit the script, remove the line with `file=`, then run...
```
:global EmailBackupCc  "mail@eworm.de";
```
... and run the script. Do not forget to revert the change for `EmailBackupCc`.

eworm-de, 2019-07-18:
Well, wondering if your SMTP server is the problem... Can you configure an account from a different SMTP server?

Kampfwurst, 2019-07-18:
If i use my server i get the ":invalid BODY" error.

I try the google server also. I can send over the google server a testmail from Tools/Email
If i try to execute the script i get the error "error talking to server"

eworm-de, 2019-07-18:
I received two mails from two different addresses.

I guess both servers do not like mails from RouterOS devices when attachments are included... You could enable logging and see if anything useful pops up there:
```
/system logging add topics=e-mail;
```

Also if one of the servers is yours... Have a look at the MTA's logs.

Anyway... I think this is out of scope for my scripts. Looks like the issue is how RouterOS assebles the attachment into the mail.

Kampfwurst, 2019-07-18:
very strange. This script is working from the same RB with the server. Also with the attachment

`:log info "Starting Backup Script..."
:global backupfile ([/system identity get name] . ".rsc")
:if ([/file find name=$backupfile] != "") do={/file rem $backupfile}
:delay 2s
/export file=$backupfile
:log info "Waiting 5s for backup to complete..."
:delay 5s
:log info "Backup being emailed..."
/tool e-mail send to="XXXX@test.de" subject=([/system identity get name] . " Backup") from=XXXX@test.de file=$backupfile server=testsmtp.de
:log info "Finished Backup Script!"`

eworm-de, 2019-07-18:
Really strange... Is it possible that the file name confuses the servers? Possibly too many dots? Replace the line for file name with just:
```
:local FileName $Identity;
```
... and try again.

Kampfwurst, 2019-07-18:
Ok now its working fine. It seems that "mikrotik" has a problem with the dots

eworm-de, 2019-07-18:
No, it's you provider that has problems with dots. ;)
Imaging someone sends emails with a file with name `holiday.png.exe`, stupid mail clients strip the file extension. How many people will click it? Looks like providers try to mitigate by denying attachments with more than one dot in file name.

commit 431a4c81 (2019-07-18): email-backup: no more than one dot in file name

Looks like some providers do not allow more than one dot in attachment's
file name to mitigate something like `holiday.png.exe`.

Let's just replace dots with underscores.

Fixes #2

Reported-by: @Kampfwurst
closed 2019-07-18 by eworm-de