eworm-de/routeros-scripts #3: Anleitung

Kampfwurst, 2019-09-01:
Hallo Christian,

Hast du eine Anleitung wie genau man die Scripte einbinden etc muss. Probiere gerade seit 2 Tage das z.B SMS Befehls Script zum laufen zu bekommen.

sG

eworm-de, 2019-09-02:
Ich muss dringend zu den ganzen Scripts Dokus schreiben... Werde mich dran machen wenn ich ein wenig Luft habe.
Zu `sms-action` habe ich noch vor ein paar Tagen eine englische Mail geschrieben, das kopiere ich mal hier rein. Kommst du damit klar?

eworm-de, 2019-09-02:
Die Basis-Scripts (und `sms-action`) hast du gemäß der Anleitung installiert?
https://git.eworm.de/cgit.cgi/routeros-scripts/about/

Zusätzlich muss der SMS-Empfang sowie der Script-Kram in `/ too sms`
konfiguriert sein. Sieht bei mir etwa so aus:

```
/tool sms set allowed-number=+49123456 auto-erase=yes port=lte receive-enabled=yes secret=v3ry-s3cr3t
```

Dann kannst du eine SMS von der Absendernummer `+49123456` mit folgendem Inhalt
an die Nummer deines Routerboards schicken:

```
:cmd v3ry-s3cr3t script sms-action action=reboot
```

Was für den action-Parameter akzeptiert wird kannst du im Script
`global-config` einstellen.

Ich hoffe damit kommst du zurecht... Zusätzliche Infos gibt es noch im Wiki:
https://wiki.mikrotik.com/wiki/Manual:Tools/Sms#Receiving

Kampfwurst, 2019-09-02:
Danke hat funktioniert. Ich wusste nicht wie ich genau den Befehl in der SMS schreiben muss. ;-) Ansonsten super Arbeit. Ich lasse dir auch was per Paypal zukommen.

closed 2019-09-02 by Kampfwurst

eworm-de, 2019-09-02:
Vielen Dank! :-D