eworm-de/routeros-scripts #4: Failed sending telegram notification

rchovan, 2020-01-28:
Hi, 
first of all, Thank You for such a great library of script. I'm using at home this for forwarding my SMS to Telegram. It was working ok, but it suddenly stopped. I have tried to update script, but I have same error in logs: 

[image]

I have noticed, that in script is used 
```
/ tool fetch check-certificate=yes-without-crl keep-result=no 
```

but according to RouterOS, it is deprecated parameter

[image]

This is my setup:
```
[rch@mikrotik] /system identity> /system routerboard print 
       routerboard: yes
        board-name: hAP ac^2
             model: RBD52G-5HacD2HnD
     serial-number: xxxxxxxxxxxx
     firmware-type: ipq4000L
  factory-firmware: 6.42.3
  current-firmware: 6.44.3
  upgrade-firmware: 6.46.2
```

Could you please modify script to work with 6.46.x version ?

Thank You

eworm-de, 2020-01-28:
I did not notice that `keep-result=no` is deprecated. Just replaced it with `output=none` in 5741924cf92f3ff5c403d7d1ea34d5d6a47c8cf7, 639011a9bac9adc2c065d497f1a9d03151e8c31e and 74f6449e8ac06e30af5511c1876ee518ae8926b9.
Thanks for the hint!

However this is not the cause of your issue, I use the script all the time with RouterOS 6.46.2.

My first guess is that something is fishy with the certificates. Can you please give the output of `/ certificate print where common-name~"Go Daddy"`?

rchovan, 2020-01-28:
Hi, 
here is output
```
[rch@mikrotik] /system identity> / certificate print where common-name~"Go Daddy"
Flags: K - private-key, L - crl, C - smart-card-key, A - authority, I - issued, R - revoked, E - expired, T - trusted 
 #         NAME                              COMMON-NAME                             SUBJECT-ALT-NAME                                                          FINGERPRINT                            
 0       T Go Daddy Secure Certificate Au... Go Daddy Root Certificate Authority ...                                                                           45140b3247eb9cc8c5b4f0d7b53091f73292...
 1  L    T Go Daddy Secure Certificate Au... Go Daddy Secure Certificate Authorit...                                                                           973a41276ffd01e027a2aad49e34c37846d3...
[rch@mikrotik] /system identity> 

```

I have tried just fetch tool with hardcoded telegram bot and chat id, but it ends with error 

```
[rch@mikrotik] /system identity> / tool fetch check-certificate=yes-without-crl http-method=post ("https://api.telegram.org/bot<big_telegra_id>/sendMessage") http-data=(d
isable_notification=false&text=skuska&chat_id=25533449) output=none                                          
  status: failed

failure: closing connection: <400 Bad Request> 149.154.167.220:443 (4)

```

eworm-de, 2020-01-28:
The certificates look ok.

Something is wrong with your Telegram settings. Either `$TelegramTokenId` or `$TelegramChatId` is incorrect. I tested with your given `$TelegramChatId` `25533449` and received an error as well. Is that the id you are trying with?

rchovan, 2020-01-28:
double checked TokenId and ChatId, it is correct. Tried with Postman, and it is working. 
But when I deleted Go daddy certificates, and then run script for sending notification, new certificates were imported, and now it is working correctly: 

```
22:24:35 script,info Certificate with CommonName "Go Daddy Secure Certificate Authority - G2" not available. 
22:24:35 script,info Downloading and importing certificate with CommonName "Go Daddy Secure Certificate Authority - G2". 
22:24:35 info fetch: file "Go Daddy Secure Certificate Authority - G2.pem" downloaded 
22:24:37 certificate,info CRL updated for Go-Daddy-Secure-Certificate-Authority-G2 
```

I don't know, if it counts, but previous certificate had different name 

[image]

Anyway, Thanks for help :-)

closed 2020-01-28 by rchovan

eworm-de, 2020-01-28:
Naming of certificates changed in b3a76c7e4b5e32be9fcffdeeecfa8c61f365de22. As the name just has a descriptive role that's not the issue. I can not explain what could have caused your issue...

Anyway... Have fun with the scripts! ;)

eworm-de, 2020-01-28:
If this happens again please have a look at free flash space and logs regarding certificates. Possibly RouterOS tries to download CRLs and fills up free space?

eworm-de, 2020-01-29:
Added a warning about CRL download with low free flash space in 801dce05fa593f61103a0e48d2175bcc58c061a0.
Does this trigger for you?

rchovan, 2020-01-29:
No, I know, that hAP AC2 is low on space, but I have still 8% (1.3MB) free
```
                   uptime: 3d12m8s
                  version: 6.46.2 (stable)
               build-time: Jan/14/2020 07:17:12
         factory-software: 6.42.3
              free-memory: 61.4MiB
             total-memory: 128.0MiB
                      cpu: ARMv7
                cpu-count: 4
            cpu-frequency: 716MHz
                 cpu-load: 0%
           free-hdd-space: 1344.0KiB
          total-hdd-space: 15.3MiB
  write-sect-since-reboot: 447674
         write-sect-total: 31610322
               bad-blocks: 0%
        architecture-name: arm
               board-name: hAP ac^2
                 platform: MikroTik
```

eworm-de, 2020-01-29:
This is not about certificate but about [certificate revocation list](https://en.wikipedia.org/wiki/Certificate_revocation_list) (crl). That's a signed list of issued certificates that have been revoked. Depending on the number of revoked certificates this list can be quite huge.