eworm-de/routeros-scripts #6: dhcp-to-dns script question.

tspinoso, 2020-07-02:
Hi,  first thank you for all your hard work on this.  I am a bit of a ROS newbie so please bear with me.

I installed the dhcp-to-dns script and am using a simple internal .lan for my domain name.  I would like something like hostname.lan.  Instead I am getting hostname.dhcp.lan.  I have the HostNameInZone configured set to false.  Any way to have this appear simply as hostname.lan?

Also - I am unclear why there is the global-config overlay VS just global-config.  Which one takes precedent?  Does one update the other?

Thanks in advance and sorry of the layman questions.

eworm-de, 2020-07-03:
> I installed the dhcp-to-dns script and am using a simple private internal .lan for my domain name. I would like the static DNS record to be hostname.lan. Instead I am getting hostname.dhcp.lan. I have the HostNameInZone configured to false, hopefully that is correct. Any way to have this appear simply as hostname.lan?

There was no way to use just `hostname.lan`. Changing `HostNameInZone` switches between `hostname.dhcp.lan` and `hostname.dhcp.identity.lan`, where `identity` is your routers hostname.

Actually there's no reason not to allow just `hostname.lan`... Possibly except this can conflict with other static hostnames.
Anyway, decided to make the configurable in fe2dc8d9b399e89570db884a1272e8139e150dcd, so update you scripts and set `PrefixInZone` to `false`.

> Also - I am unclear why there is the global-config overlay VS just global-config. Which one takes precedent? Does one update the other?

The `global-config` is updated form my repository (so manual changes are overwritten!), this makes sure upstream changes make its way to the user. The script `global-config-overlay` takes precedence and is the one a user should edit.

closed 2020-07-03 by eworm-de

tspinoso, 2020-07-03:
Thank you for the update and your explanation.

eworm-de, 2020-07-03:
You are welcome!
And do not forget to leave your star... ;)

kpoleyn, 2021-11-21:
Nice work! Just got it working, added a few other scripts. 
Great work, it would've taking me ages to compile something like this (with all the features).

Thank you!