eworm-de/routeros-scripts #7: Netwatch-Notify configuration questions.

tspinoso, 2020-07-03:
Hi, 
Again - thank you for all of this - I am leaning much about how scripting works and I sincerely thank you.

If this is not the best way to ask questions, as these are not code issues necessarily, please let me know.

Questions - I am looking through this and trying to understand the "several" checks aspect of this script.  If I understand correctly the default if =>5?  So if the time out is 5s then it would roughly check for a period of 25 seconds before initiating the email?

Also - if I was going to turn an interface on/off or enable/disable a firewall rule would I add them as I have below in my example as SET CUSTOM ACTIONS HERE?

` :if ($Metric->"count" >= 5 && $Metric->"notified" != true) do={
      $SendNotification ("Netwatch Notify: \E2\9D\8C " . $HostName . " down") \
        ("Host " . $HostName . " (" . $HostVal->"host" . ") is down since " . $HostVal->"since" . ".");
      :set ($Metric->"notified") true;
      **SET CUSTOM ACTIONS HERE**
    }`

Many thanks again.

eworm-de, 2020-07-03:
> Again - thank you for all of this - I am leaning much about how scripting works and I sincerely thank you.

You are welcome. ;)

> If this is not the best way to ask questions, as these are not code issues necessarily, please let me know.

This is open for whatever kind of discussion. Github decided to call it "issues". :-p

> Questions - I am looking through this and trying to understand the "several" checks aspect of this script. If I understand correctly the default if =>5? So if the time out is 5s then it would roughly check for a period of 25 seconds before initiating the email?

Correct.

> Also - if I was going to turn an interface on/off or enable/disable a firewall rule would I add them as I have below in my example as SET CUSTOM ACTIONS HERE?

Yes.

Wondering if adding support for hooks makes sense... Something like `up-hook=/ interface disable ..., down-hook=...`. This would work exactly like netwatch's `up-script` and `down-script`, but with the state machine (several checks) on top.

tspinoso, 2020-07-03:
> https://forum.mikrotik.com/viewtopic.php?f=9&t=156466

tspinoso, 2020-07-03:
> Wondering` if adding support for hooks makes sense... Something like up-hook=/ interface disable ..., down-hook=.... This would work exactly like netwatch's up-script and down-script, but with the state machine (several checks) on top.

I think that would be very helpful and a much welcomed enhancement.

p.s. Sorry about the open/closed thing.  Still trying to get the hang of this.  170th star left :-)

eworm-de, 2020-07-03:
Please have a look at 1c10044b2430c4bd8ec5955ead84fbe282587bef (or update and try it...).

Only `down-hook` is available, as `up-hook` would do the same as netwatch's `up-script`.

eworm-de, 2020-07-03:
Why did you post the link to _Mikrotik RouterOS automatic backup and update script_?
I have similar scripts in my repository.

tspinoso, 2020-07-03:
> Please have a look at [1c10044](https://github.com/eworm-de/routeros-scripts/commit/1c10044b2430c4bd8ec5955ead84fbe282587bef) (or update and try it...).
> 
> Only `down-hook` is available, as `up-hook` would do the same as netwatch's `up-script`.

I will - thank you!

tspinoso, 2020-07-03:
> Why did you post the link to _Mikrotik RouterOS automatic backup and update script_?
> I have similar scripts in my repository.

I am not sure what you mean?  I posted it in here?  I apologize and did not mean to please remove it.

eworm-de, 2020-07-03:
It's gone, so I guess you removed it yourself.

If my change suits your needs feel free to close, else complain. ;)

tspinoso, 2020-07-03:
Ah - now I get it. You saw it in-between the time I mistakenly posted it and removed it.  Embarrassingly, I was trying to figure out how to quote your post and tried to copy and paste.  That link was in my clipboard so it pasted instead and posted the comment.... then I hit close and comment and it went downhill from there :-(.  All fixed now I hope and I apologize.  

All that said - I have some unique thoughts in general about how these _"general backup and update"_ scripts work.  None seem to fit what I am looking for in my use case.  If you are interested let me know and I would be happy to share, perhaps my intentions would have some value for what you have.  To be clear, I am not 100% certain your script does not already accomplish what I am after, but I will do my due diligence soon.  Perhaps a conversation better had offline?

Certainly no complaints from me!  This is perfect.  I will reply back if I have issues.  Thanks for all of your hard work.  I owe you a donation for your work.  you can count on it when I can.

Closing now....

closed 2020-07-03 by tspinoso

eworm-de, 2020-07-03:
> All that said - I have some unique thoughts in general about how these "general backup and update" scripts work. [...]

Feel free to open a new issue or contact me via mail, Telegram or whatever.

tspinoso, 2020-07-03:
Thank you sir - I will post an email to you later after I have given yours  solution careful consideration.  I welcome your thoughts on my update/backup needs.