eworm-de/routeros-scripts #11: ipv6-update adds 1 to the static IPv6 DNS-name's address if it ends in even number

peeterrussak, 2021-05-23:
Tested on RouterOS v7.1b6 Example prefix: 2001:7d0:87cc:c600::/56

1. Create static IPv6 DNS record ending with ::1
2. Disable dhcpv6-client, enable dhcpv6-client -> new prefix propagates correctly, static address ends with ::1
3. Create static IPv6 DNS record ending with ::2
4. Disable dhcpv6-client, enable dhcpv6-client -> new prefix propagates correctly, but address ends with ::3
5. Disable dhcpv6-client, enable dhcpv6-client -> new prefix propagates correctly, static address ends correctly with ::3

eworm-de, 2021-05-24:
I have not seen this... Will try to reproduce later today.
Can you please give exact commands to create the DNS records?

eworm-de, 2021-05-24:
BTW... I did not test my scripts with RouterOS 7.x at all... Possibly they broke arithmetics with IPv6 addresses? I will check.

peeterrussak, 2021-05-24:
Re-tested with 6.48.2 and got same result. And figured it out. This one doesn't get prefix, but IPv6-address of interface:
    :local Prefix [ / ipv6 address get [ find where interface=($Comment->"interface") from-pool=$Pool global ] address ];

If interface's address is smth else than :: then it's OR-ed to the static address. I had ::1, which explains +1: 

Let's test:
/ipv6 address add address=::100 from-pool=isp interface=bridge
/ip dns static add address=::2 comment="isp-pool-isp, interface=bridge" name=test.test ttl=15m;
/ipv6 dhcp-client disable ether1
/ipv6 dhcp-client enable ether1

With added logging:
    :log info ($Prefix);
    :set Prefix [ :toip6 [ :pick $Prefix 0 [ :find $Prefix "/64" ] ] ];
    :log info ($Prefix);
    :log info ("Address is: ". ($RecordVal->"address"));
    :local Address ($Prefix | ([ :toip6 ($RecordVal->"address") ] & ::ffff:ffff:ffff:ffff));
    :log info ("Updating DNS record for " . ($RecordVal->"name") . ($RecordVal->"regexp") . " to " . $Address);

Creates output:
21:53:29 system,info static dns entry added by admin 
21:53:34 system,info DHCPv6 client on ether1 changed by admin 
21:53:36 system,info DHCPv6 client on ether1 changed by admin 
21:53:37 script,info Updating IPv6 address list with new IPv6 prefix 2001:7d0:87c6:7b00::/56 
21:53:37 system,info address list entry changed by admin 
21:53:39 script,info 2001:7d0:87c6:7b00::100/64 
21:53:39 script,info 2001:7d0:87c6:7b00::100 
21:53:39 script,info Address is: ::2 
21:53:39 script,info Updating DNS record for test.test to 2001:7d0:87c6:7b00::102

eworm-de, 2021-05-24:
Ah, got it... Can you please test?

```
/ system script set comment="\?h=next" ipv6-update
$ScriptInstallUpdate
```

Also fix the address in dns record, then disable and re-enable.

closed 2021-05-26 by eworm-de

eworm-de, 2021-05-26:
Already merged to `main` branch... Please complain if your issue remains.
... and leave a star if you have not.

jojoamrni, 2021-05-26:
Please how can I write a script like you guys? I'm so back in the world of technology.... I use Mikrotik, but want to be a GURU in it that I can set up a network and secure it.

eworm-de, 2021-05-26:
Please do not hijack issues with nonsense topics.

peeterrussak, 2021-06-02:
Thanks, works now as expected!

commit e74b2e03 (2021-05-24): ipv6-update: apply a mask on prefix

Fixes #11