eworm-de/routeros-scripts #14: :global MkDir;

sunnychuchu, 2021-06-21:
Hi,

I have deployed about 10 pcs of router for different models , some of router is unable create email-backup folder when first run of script.   [image]
Once folder is created, all email-backup will run successfully on 2nd times and so on.

Sunny

eworm-de, 2021-06-21:
Hmm, I have not see this to fail so far...

Ok, let's start with some details: RouterOS does not provide a lot of functionality regarding file handling. Creating directories is not implemented for example, so I have to use a workaround. That workaround is to enable the `www` service to listen on loopback address, then download a file to a path including the directory, finally removing the file which leaves the directory.

So I wonder...
* Did you have any special configuration for the `www` service?
* Please show output of `/ip service export`.
* Are there any firewall rules that block connections to loopback address?
* Also please check if any settings are altered (enabled with address `127.0.0.1/32` on port `80`) but not restored to the original settings.
* The second attempt succeeds or do you create the directory manually?
* Do the affected devices use a ram disk? Please show output of `/file print count-only where name="flash" type="disk"`.

eworm-de, 2021-06-21:
Let's try something new, which should work with less trouble.
Can you try the changes in a424691a68dd2d0c12a979e58a5c95a21bdee1de (or `next` branch)?

sunnychuchu, 2021-06-22:
It seem much better after using SMB method. I will perform more test on different model of MTK router. 
Thanks for your advancement 
Sunny

eworm-de, 2021-06-22:
Just merged this into the `main` branch.
It works for me™️ and - though it is still a workaround - looks cleaner than what we had before. Thanks for the report!

commit a424691a (2021-06-21): global-functions: $MkDir: abuse smb share to create directory

The smb feature is provided by system package... So we have it anyway.
It gives some benefits compared to abusing fetch:

* It is faster!
* No need to alter, enable and restore a service! (The share is created disabled.)
* Firewall rules can not break this.
* No temporary file is created.
* Less code!

Let's hope we do not introduce new breakage.

Closes #14
commit 7fb4fdfc (2021-06-21): global-functions: $MkDir: abuse smb share to create directory

The smb feature is provided by system package... So we have it anyway.
It gives some benefits compared to abusing fetch:

* It is faster!
* No need to alter, enable and restore a service! (The share is created disabled.)
* Firewall rules can not break this.
* No temporary file is created.
* Less code!

Let's hope we do not introduce new breakage.

Closes #14
closed 2021-06-22 by eworm-de