eworm-de/routeros-scripts #17: Netwatch notify number of checks before wan-up?

dragon2611, 2021-07-26:
On the netwatch notify Is it possible to use the count for wan-up as well as wan-down? (or have a separate count for WAN-Up)

Ideally I'd like the wan to pass a certain number of checks before it's brought back into use to deal with a link that's not completely dead but has high levels of packetloss.  (E.g a wireless link in bad weather)

eworm-de, 2021-07-27:
Currently this is not possible...

Wondering if it makes sense. Note that netwatch only checks every interval. So a flapping link that is down most of the time but passes a ping at interval is considered up anyway... This would be unreliable at best.

Let me think about this...

dragon2611, 2021-07-27:
You would have to increment a counter similar to how it's done for links that are down I guess.