eworm-de/routeros-scripts #26: How use script s with Netwatch ?

Jason-MK, 2022-07-08:
Hello ,

I want to know how to use the scripts with Netwatch ?

I mean should I write the script ? there are no free scripts ?

I mean I want to have a script to send me a message where any GRE Tunnel is down for 1 second ?

I really need your help

Best Regards

eworm-de, 2022-07-08:
Wait... What?! You want to act if a GRE tunnel is down for just a single second?
Not sure this is a good idea...

I have a scripts to [Notify on host up and down](https://git.eworm.de/cgit/routeros-scripts/about/doc/netwatch-notify.md) (and optionally run hooks/scripts), but the basic idea is to have a state machine to delay actions and mitigate false positives.

As I have nothing appropriate and you already [opened a thread on forums](https://forum.mikrotik.com/viewtopic.php?t=187461)... Closing.

closed 2022-07-08 by eworm-de

Jason-MK, 2022-07-09:
> 

Hello @eworm-de 

I want to know just an answer for a simple question , 

So all what I need to install the script or copy it to the scripts but then how do I added one of the hosts ? 

best regards

eworm-de, 2022-07-10:
You need to read the documentation.

taahgraham, 2022-10-11:
You would be looking for a simple icmp script, point it at the default
gateway of remote end-point 👍