eworm-de/routeros-scripts #27: Autoreboot?

dima1002, 2022-08-20:
With this script, the router should restart after 1 minute if autoupgrade is switched on and an upgrade is available, right? Because nothing happens to me.

eworm-de, 2022-08-20:
You are speaking about what? `firmware-upgrade-reboot`?
In fact it does not matter if auto-upgrade is enabled. Just run the script it will upgrade and reboot as required.

dima1002, 2022-08-20:
Yes exactly from the script.
After a packages update, a firmware update must be made and then a 2nd reboot. Is not that so?

eworm-de, 2022-08-20:
The firmware update is not strictly required (unless mentioned in changelog), but I do update and keep the versions in sync just to be sure. So what's the output of `/system/routerboard/print`?

eworm-de, 2022-08-20:
Also the script should give some output. What happens if you run it?
```
[admin@MikroTik] > /system/script/run firmware-upgrade-reboot 
info: Firmware is already up to date.
```

dima1002, 2022-08-20:
I want to keep it in sync too.
The Mikrotik access point should always reboot automatically when the following message appears in the log: "Firmware upgraded successfully, please reboot for changes to take effect!"

Here is my output:
[admin@MikroTik] > /system/routerboard/print 
             ```
   ;;; Firmware upgraded successfully, please reboot for changes 
                    to take effect!
       routerboard: yes
        board-name: hAP ac
             model: RB962UiGS-5HacT2HnT
          revision: r2
     serial-number: F7D11F682D27
     firmware-type: qca9550L
  factory-firmware: 6.44.6
  current-firmware: 7.1.3
  upgrade-firmware: 7.4.1

```

dima1002, 2022-08-20:
Unfortunately nothing happens. The output hangs
[image]

eworm-de, 2022-08-20:
The documentation starts with:

> ℹ️ Info: This script can not be used on its own but requires the base installation. See [main README](https://git.eworm.de/cgit/routeros-scripts/about/README.md) for details.

You should do that.

closed 2022-08-20 by eworm-de