eworm-de/routeros-scripts #29: Expect to have exponentiation and file movement capabilities.

wy16W2pIilK1xgqN, 2022-09-29:
Scripts lack the ability to exponentiation operations and file moves.
Hope you can achieve it.

eworm-de, 2022-09-29:
What is *exponentiation operations*?

File move is not supported by RouterOS. Use a SCP/SFTP client...

wy16W2pIilK1xgqN, 2022-09-30:
Sorry, didn't make it clear.
I don't know English and use translation software to communicate with you.

The mathematical formula goes like this.
```
2⁴ = 16

2⁸ = 256

4⁴ = 256

```

This is true in linux bash.

```
root@archlinux:~#  echo $[2**4]
16
root@archlinux:~#  echo $[2**8]
256
root@archlinux:~#  echo $[4**4]
256
```

eworm-de, 2022-09-30:
You can use a simple function for that:
```
:global Exp do={
  :if ($2 = 0) do={ :return 1; }
  :local Return 1;
  :for I from=1 to=$2 do={
    :set Return ($Return * $1);
  }
  :return $Return;
}
```
... store it as script `mod/exp` if you want to make it permanent.

To use the function:
```
[admin@MikroTik] > :put [ $Exp 2 4 ]
16
[admin@MikroTik] > :put [ $Exp 2 8 ]
256
[admin@MikroTik] > :put [ $Exp 4 4 ]
256
```
I do not have a use case for this in my scripts, so not adding.

closed 2022-09-30 by eworm-de