eworm-de/routeros-scripts #33: Foreign scripts & modules

eworm-de, 2022-12-05:
Hello everybody,

people start to contribute scripts, but I am not interested in using all of them, or even missing hardware to do so. Merging them here is a bad trade off for both of use, the contributor and me: Issues are opened here, the contributor does not recognize (without me forwarding them). On the other hand all changes would have to be reviewed and merged by me, mostly without testing.

To mitigate that situation I think about adding a new section [Foreign scripts & modules](https://github.com/eworm-de/routeros-scripts/tree/foreign-scripts-modules#foreign-scripts--modules) to the main README, just below [Installing custom scripts & modules](https://github.com/eworm-de/routeros-scripts#installing-custom-scripts--modules). You could add links to your website or repository with the scripts, and link back to my repository for general information, base installation and other scripts.

Any comments on this? I would like some feedback from @zainarbani (for #22) and @bubenkoff (#31 & #32). Do you think this is a reasonable solution for you?

bubenkoff, 2022-12-06:
IMO it's a bit strange concept of 'foreign' vs 'native' scripts.
Really, they are all 'foreign' to routers, hence the collection was created
in the first place.
If there's a concern with testing - let's work on that, there should be a
way to implement automated testing using the virtual device, or at least
some basic script validation.
Those who contribute scripts should IMO become members of the repository so
that they can handle potential user issues, so testing is
naturally resolved then, even without automated testing.

zainarbani, 2022-12-07:
LGTM, people can always fetch their custom/self-made script with `$ScriptInstallUpdate`.
Create dedicated repos for custom scripts also a good idea, thus will make it easier to maintain main global _functions/hooks/converters_ here.
Migration to full `routeros-scripts-helper` that focuses on _functions/hooks/converters_ docs.
Again, its all up-to you as main devs here.
A PR's doesn't mean that you have to merge, it can be a bug-fixes, patch, or idea that might be helpful on future.

eworm-de, 2022-12-19:
Well, I do not have a strong opinion on the name. Any suggestion is welcome, perhaps we can find a name that suits better. I will happily use that then.

Given we merge this... Does it make sense to provide a skeleton repository for people to clone/fork and start with?

eworm-de, 2023-01-22:
We have this as [Linked custom scripts & modules](https://github.com/eworm-de/routeros-scripts#linked-custom-scripts--modules) now.

closed 2023-01-22 by eworm-de