eworm-de/routeros-scripts #37: Dhcp2dns

DunkanAidaho, 2023-02-11:
Hi, first thank you for all your hard work on this. 
I use this script, but every time it runs, the processor is 100% utilized.  Tried to run from the console, the script runs for 20-30 seconds.  There are no more than 10 reservations in dhcp.
 Also, on the firewall, I redirect all traffic on port 53 from the local network to the router itself.
 So I use DoH (adguard).
 What could be the reason for the high utilization rate?

eworm-de, 2023-02-11:
Oh, it should not take that long... Will have a look.
What device is this running on? Of cause it depend on available CPU power.

DunkanAidaho, 2023-02-11:
Problem device - RB2011UiAS.
On RBD53iG-5HacD2HnD - ok

eworm-de, 2023-02-12:
Well, that device is quite limited in processing power... It has a single core MIPSBE CPU running at 650MHz.

Anyway, let's try to get some data... Please run these command:
```
/system/script/set comment="use-suffix=\?h=debug" dhcp-to-dns;
$ScriptInstallUpdate;
```

... then run the script from terminal and paste the output here.

At what load is the device running if no script is executed? Or what average throughput does it handle?

Are there any wireless devices that are constantly connecting and disconnecting, causing a lease to be bond and unbound every time?

DunkanAidaho, 2023-02-12:
`liet-phone@MikroTik-CCC] > /system/script/set comment="use-suffix=\?h=debug" dhcp-to-dns;$ScriptInstallUpdate;
info: Updating script: global-functions
info: Reloading global functions.
info: The configuration version on MikroTik-CCC increased to 94, current configuration may need modification. Please review and update global-config-overlay, then re-run global-config.
info: Change 94: Added support for host addresses in address-list for 'ipv6-update'.
[liet-phone@MikroTik-CCC] > /system/script/run
check-certificates     global-config-overlay
check-routeros-update  global-functions
dhcp-to-dns            lease-script
global-config          number
[liet-phone@MikroTik-CCC] > /system/script/run
=
[liet-phone@MikroTik-CCC] > /system/script/run dhcp-to-dns
[liet-phone@MikroTik-CCC] >`
Listing is here.
With this version worked for 2 seconds.  The load did not exceed 40%

 Under load, the processor is utilized during the day by 50-60% on average.  (According to the Prometheus extractor)

 Of course, there are wireless devices on the network.  15-20% of the entire pool of given addresses (40-50) .  but even when DHCP lease passes and there is still no device on the network.  Script took a long time to run

eworm-de, 2023-02-12:
The script has not been modified at all for you. I guess you copied the commands from mail? There was a typo, that I fixed in comment above. Sorry for the confusion. Please try again, and make sure the comment contains `url-suffix` (not `use-suffix`).

DunkanAidaho, 2023-02-12:
et-phone@MikroTik-CCC] > /system/script/run dhcp-to-dns                                            Starting: 12:48:57
Running: 12:48:57
Records: 0
Leases:  16
Step: 12:48:58
Done: 12:49:00

eworm-de, 2023-02-12:
It is running for three seconds... Which sound kind of ok for that device.

Please update again, I did another little change. Then try to reproduce the problem where it takes 20 to 30 seconds.

DunkanAidaho, 2023-02-12:
Thank you!  I can only check tomorrow when the device is under more load.  Today is a day off.  I will notice you

DunkanAidaho, 2023-02-13:
На утро результат следующий...
`to-dns                                            Starting: 09:41:04
interrupted
[liet-phone@MikroTik-CCC] > /system/script/set comment="url-suffix=\?h=debug" dhcp-to-dns;$ScriptInstallUpdate;
[liet-phone@MikroTik-CCC] > /system/script/run dhcp-to-dns                                            Starting: 09:42:22
info: Script 'dhcp-to-dns' started more than once and timed out waiting for lock... Aborting.
Hard error to exit.
[liet-phone@MikroTik-CCC] > /system/script/run dhcp-to-dns
Starting: 09:43:05
error: More tickets than running scripts 'dhcp-to-dns', resetting!
interrupted
           interrupted
[liet-phone@MikroTik-CCC] > /system/script/run dhcp-to-dns
Starting: 09:43:15
Running: 09:43:16
Records: 50
Leases:  54
Step: 09:43:35
Done: 09:45:01`

eworm-de, 2023-02-17:
Ok, let's see step by step...
```
[...]to-dns
Starting: 09:41:04
interrupted
```
Well, that should not happen. And usually it does not. It does happens on resource shortage, for example no more free RAM.
```
[liet-phone@MikroTik-CCC] > /system/script/set comment="url-suffix=\?h=debug" dhcp-to-dns;$ScriptInstallUpdate;
[liet-phone@MikroTik-CCC] > /system/script/run dhcp-to-dns
Starting: 09:42:22
info: Script 'dhcp-to-dns' started more than once and timed out waiting for lock... Aborting.
Hard error to exit.
```
The script is not allow to run in parallel. So this is expected when run twice.
```
[liet-phone@MikroTik-CCC] > /system/script/run dhcp-to-dns
Starting: 09:43:05
error: More tickets than running scripts 'dhcp-to-dns', resetting!
interrupted
```
This is probably due to an interrupted run, like the one above.
```
[liet-phone@MikroTik-CCC] > /system/script/run dhcp-to-dns
Starting: 09:43:15
Running: 09:43:16
Records: 50
Leases:  54
Step: 09:43:35
Done: 09:45:01
```
Ok, it is really runtime, not just waiting for what ever. But there is nothing I could do about it. Checked the script, no hidden penalties. I guess the device is just low on resources, especially spare CPU cycles.

So probably best is to upgrade the hardware... Something like RB4011 or RB5009 could do well, or even hAP with ARM CPU.

If you still want to keep the device but can live with a delay on setting the DNS name... Drop the script from DHCP servers, and run it from scheduler:

```
/ip/dhcp-server/set lease-script="" [ find ];
/system/scheduler/add interval=15m name=dhcp-to-dns on-event="/system/script/run dhcp-to-dns;" start-time=startup;
```

closed 2023-02-17 by eworm-de