eworm-de/routeros-scripts #39: Feature Request: dhcp-to-dns - use Access List comment and/or ignore if already in DNS

johnmmcgee, 2023-03-20:
Hello,

I would like to request 2 features for the dhcp-to-dns script.

1) Allow an option to use the comment that has been set in the CapsMan Access List to be set as the hostname in dns
2) if the DNS entry already exist for the IP, do not add another entry.

Thank you

johnmmcgee, 2023-03-20:
Actually, since the comment from the access list is already being brought over to the DHCP Leases, perhaps things could simplified and just have an option to use the comment from DHCP.

eworm-de, 2023-03-21:
Comments from Telegram Chat, in case it has some more important details:

> DHCP comments for dns names would be a nice addition, as it would be in sync with using access list comments for dhcp entries and it would also remove issues with multiple entries and some items reporting information wrong, such as using blocky as a dns proxy.  If i have multiple entries, when i look at the statistics it will show one or the other.  I am open to discussion about any reason this should not be a good feature request.
> also it would remove some of the inconsistencies i have come across.  for example, at times phone entries will go in as mac address naming into DNS. etc
> I'd be willing to donate a case of beer for the cause :D
> or perhaps i could attempt to write it myself and contribute.

eworm-de, 2023-03-21:
Well, all the _DHCP leases_ and _DNS static records_ stuff is kind of tricky. I am happy that it kind of works currently. 😜

There are some issues with this:

- Currently the order is: `dhcp-to-dns` -> `collect-wireless-mac` -> `dhcp-lease-comment`... This make sense as I have the DNS name in place to add it into the notification from `collect-wireless-mac`. To get the info into `dhcp-to-dns` I would have to switch order - which causes other troubles - or try to get the info from access list directly. The latter would require to duplicate the script for local wireless and caps-man.
- The comment in access list is not limited in what it is allowed to contain. Actually it is intended to contain human readable information. Squeezing that into valid DNS names could be troublesome. 😝
- Probably more I do not recognize now.

However I do agree that it is inconsistent to have the mac address for some and host name (if available) for other records. So I am preparing some changes now. The plan is to always have an `A` type records with the mac address, and a `CNAME` type records pointing to the `A` record if the host name is available. (To test this install `dhcp-to-dns` from `next` branch.)

So wondering if you could just create another `CNAME` type record that suits your needs. It can always point to the `A` type record with the mac address then, and it resolves correctly even if the ip address changes.

eworm-de, 2023-03-21:
Also could you give some more detail on the issue with blocky dns proxy and the statistics thing that shows _one or the other_? Or is that solved by using `CNAME` type records?

johnmmcgee, 2023-03-21:
> However I do agree that it is inconsistent to have the mac address for some and host name (if available) for other records. So I am preparing some changes now. The plan is to always have an `A` type [...]

Could an option be created that would do a CNAME record that is pointing to the A record if the hostname OR a comment is available?  I would imagine you could make an option switch in the global-config that would let you ask a preference of which one.

Either hostname OR comment, or prefer comment over hostname or something along those lines. 

> Also could you give some more detail on the issue with blocky dns proxy and the statistics thing that shows _one or the other_? Or is that solved by using `CNAME` type records?

Not certain why this does this.  I would have to look further into way blocky operates.

eworm-de, 2023-03-22:
The issues from above would still apply, no?
Thinking about it, but as soon as the comment from access list comes into the game things become a lot more complex.

johnmmcgee, 2023-03-22:
I suppose it isnt that complicated from my perspective as my access list comments are pretty simple.  I set the comment to the hostname or purpose.  Example my kids laptop is named link, so i just comment it as link.  this carries over to the dhcp server comments.  the option would be nice to allow that comment to be placed in as the dns entry as well.

I will make an attempt to mock something up.

eworm-de, 2023-03-25:
You should not spread the discussion over at least three places... Let's keep it here.

I gave it some more thought and came up with 7d5c839f32bf86653e9da3051fa0ea1a53334d6c... It does not use plain comment, but would require you to set `hostname=link` for your kid's laptop.

Also note that this requires a second execution of the script with the default order of lease scripts if the comment comes from access list.

johnmmcgee, 2023-03-26:
No worries @eworm-de.  My intention of messaging you over the Telegram chat was not to fragment the conversation.  I had some questions about scripting under routeros, and I assumed that would be a quicker way to chat.  My intention was to share my work here.

I rather like your solution.   I can just put `hostname=link" in the access list and it would eventually carry over, it looks like.

Here is what I came up with today.  You can use the modified dhcp-to-dns.rsc below by setting the following in global-config-overlay

`UseDHCPConfig true;`

How this will work is it will first use a comment from the dhcp server, if available.  If not it will use the hostname and lastly if neither is available it will create an A record with the mac address. 

Setting that will also forego the creation of the CNAME.

This is just a quick hack to demonstrate the functionality that I had in mind.  

dns-to-dhcp;  
```
#!rsc by RouterOS
# RouterOS script: dhcp-to-dns
# Copyright (c) 2013-2023 Christian Hesse <mail@eworm.de>
# https://git.eworm.de/cgit/routeros-scripts/about/COPYING.md
#
# provides: lease-script, order=20
#
# check DHCP leases and add/remove/update DNS entries
# https://git.eworm.de/cgit/routeros-scripts/about/doc/dhcp-to-dns.md

:local 0 "dhcp-to-dns";
:global GlobalFunctionsReady;
:while ($GlobalFunctionsReady != true) do={ :delay 500ms; }

:global Domain;
:global HostNameInZone;
:global Identity;
:global PrefixInZone;
:global ServerNameInZone;
:global UseDHCPComment;

:global CharacterReplace;
:global IfThenElse;
:global LogPrintExit2;
:global ScriptLock;

$ScriptLock $0 false 10;

:local Zone \
  ([ $IfThenElse ($PrefixInZone = true) "dhcp." ] . \
   [ $IfThenElse ($HostNameInZone = true) ($Identity . ".") ] . $Domain);
:local Ttl 5m;
:local CommentPrefix ("managed by " . $0 . " for ");
:local CommentString ("--- " . $0 . " above ---");

:if ([ :len [ /ip/dns/static/find where (name=$CommentString or (comment=$CommentString and name=-)) type=NXDOMAIN disabled ] ] = 0) do={
  /ip/dns/static/add name=$CommentString type=NXDOMAIN disabled=yes;
  $LogPrintExit2 warning $0 ("Added disabled static dns record with name '" . $CommentString . "'.") false;
}
:local PlaceBefore ([ /ip/dns/static/find where (name=$CommentString or (comment=$CommentString and name=-)) type=NXDOMAIN disabled ]->0);

:foreach DnsRecord in=[ /ip/dns/static/find where comment~("^" . $CommentPrefix) !(type=CNAME) ] do={
  :local DnsRecordVal [ /ip/dns/static/get $DnsRecord ];
  :local MacAddress [ $CharacterReplace ($DnsRecordVal->"comment") $CommentPrefix "" ];
  :if ([ :len [ /ip/dhcp-server/lease/find where mac-address=$MacAddress address=($DnsRecordVal->"address") status=bound ] ] > 0) do={
    $LogPrintExit2 debug $0 ("Lease for " . $MacAddress . " (" . $DnsRecordVal->"name" . ") still exists. Not deleting DNS entry.") false;
  } else={
    :local Found false;
    $LogPrintExit2 info $0 ("Lease expired for " . $MacAddress . " (" . $DnsRecordVal->"name" . "), deleting DNS entry.") false;
    /ip/dns/static/remove $DnsRecord;
    /ip/dns/static/remove [ find where type=CNAME cname=($DnsRecordVal->"name") comment=($DnsRecordVal->"comment") ];
  }
}

[... 26 lines cut ...]

      :if ($DnsIp = $LeaseVal->"address") do={
        $LogPrintExit2 debug $0 ("DNS entry for " . ($HostnameSet . "." . $Domain) . " does not need updating.") false;
      } else={
        $LogPrintExit2 info $0 ("Replacing DNS entry for " . ($HostnameSet . "." . $Domain) . ", new address is " . $LeaseVal->"address" . ".") false;
        /ip/dns/static/set address=($LeaseVal->"address") $DnsRecord;
      }

      :local Cname [ /ip/dns/static/find where type=CNAME cname=($HostnameSet . "." . $Domain) comment=$Comment ];
      :if ([ :len $Cname ] > 0 && [ /ip/dns/static/get $Cname name ] != ($HostName . "." . $Domain)) do={
        $LogPrintExit2 info $0 ("Host name changed, updating CNAME (pointing to " . ($HostnameSet . "." . $Domain) . ") to " . ($HostName . "." . $Domain) . ".") false;
        /ip/dns/static/set name=($HostName . "." . $Domain) $Cname;
      }
    } else={
      $LogPrintExit2 info $0 ("Adding new DNS entry for " . ($HostnameSet . "." . $Domain) . ", address is " . $LeaseVal->"address" . ".") false;
      /ip/dns/static/add name=($HostnameSet . "." . $Domain) type=A address=($LeaseVal->"address") ttl=$Ttl comment=$Comment place-before=$PlaceBefore;
      :if ([ :len $HostName ] > 0 && $UseDHCPComment = false) do={
        /ip/dns/static/add name=($HostName . "." . $Domain) type=CNAME cname=($HostnameSet . "." . $Domain) ttl=$Ttl comment=$Comment place-before=$PlaceBefore;
      }
    }
  } else={
    $LogPrintExit2 debug $0 ("No address available... Ignoring.") false;
  }
}
```

eworm-de, 2023-03-28:
Having a name based on the mac address is the most reliable we can have here, and we should not dismiss this. Let's use a `CNAME` pointing to the mac address based name for everything else - coming from host name or comment.

So no, I will not accept your changes.

But I think d700dbc00e6e22ea9521d3b359f0d3635d9af682 comes pretty close to what you requested and should serve you well.