eworm-de/routeros-scripts #43: dhcp-to-dns crashes with IP pool instead of address

tomaskovacik, 2023-06-08:
Hello, is there any nice solution on how to use "active address" instead of address? I have a bunch of leases where the address is a pool and of course, the script will crash on it because dhcp_pool0 is not valid ipv4 or ipv6. thx for the response.

tomaskovacik, 2023-06-08:
using winbox: create a new static lease in IP->DHCP->LEASES-> hit "+", in the popup window, next to the address field, select the arrow pointing down and select any of your IP pools add mac address .... 

obviously it is feature of mikrotik, anyway,  here is my lease details:

 ;;; Peter, hostname=peter
     address=dhcp_pool0 mac-address=xx:xx:xx:xx:xx:xx address-lists="" server=dhcp1 dhcp-option="" status=bound expires-after=19h57m36s last-seen=4h2m24s active-address=10.0.3.31 active-mac-address=xx:xx:xx:xx:xx:xx 
     active-client-id="y:yy:yy:yy:yy:yy:y" active-server=dhcp1 host-name="peter" 

/system/script/run dhcp-to-dns;
info: Adding new DNS entry for xx:xx:xx:xx:xx:xx (XX-XX-XX-XX-XX-XX.example.com -> dhcp_pool0).
invalid value for argument address:
    invalid value for argument ip
    invalid value for argument ipv6

eworm-de, 2023-06-08:
Ah, interesting... Did did not know this is possible.

The fix is obvious and easy, though... Just using the other property.

I've prepared the fix in `next` branch. Please try:
```
/system/script/set dhcp-to-dns comment="url-suffix=?h=next";

tomaskovacik, 2023-06-11:
Sorry for not checking it sooner, I forget :( 

But the fix works like a charm!

Many thx!

closed 2023-06-11 by tomaskovacik

eworm-de, 2023-06-11:
Thanks for testing! Just moved to code into the `main` branch, so you can now drop the comment from script:
```
/system/script/set dhcp-to-dns comment="";
```

Please leave a star ⭐️ if you have not yet. 😊