eworm-de/routeros-scripts #48: DHCP-to-DNS script IPv6 support

KuxaBeast, 2023-10-21:
I have used a similar script to `dhcp-to-dns` for assigning static DNS records to IPv4 addresses and previously modified it to also generate records for IPv6 addresses based on DHCPv4 leases' MAC address and IPv6 Neighbor List entries. I haven't thought about it much before, so there might be some scenario where this kind of solution misbehaves. Nevertheless, I believe we deserve some way to implement DNS auto-configuration for IPv6 as well...

Is this feature wanted or is the `dhcp-to-dns` script considered feature-complete? Would perhaps a separate script implementing this feature be welcome or does this feel too _hacky_?

eworm-de, 2023-10-27:
The scripts are never feature-complete... I am constantly improving all of them.

In fact I have thought about adding support for IPv6 / AAAA records in `dhcp-to-dns` several time, but always relinquished.

Let's face the details: The A records for IPv4 addresses generated from DHCP work pretty well. Generating AAAA records on top of that with a pretty fragile mechanism is a bad idea. It does not bring any benefit (as A records a available anyway), but brings extra layers of complexity. I am inclined not to accept anything like this.

Or can you give an example where this actually does bring a real benefit?

KuxaBeast, 2023-10-27:
Well, when I build my network on dual-stack, I don't think IPv4 records are enough for me. I consider the possibility of having devices in remote IPv6-only networks (on VPN or other medium) trying to access my local network and in case of having just A records, they would not resolve my devices. I actually have a practical use case with a WireGuard-based OSPF2/3 network and since NAT64/46 is kinda not possible on ROS atm I can't really cheat that :D

eworm-de, 2023-10-27:
But for remote IPv6-only devices you do not have IPv4 DHCP leases that you could derive the AAAA records from, no?

eworm-de, 2023-10-27:
Ah, wait... Your case is the other way round? Your remote devices should access local devices via IPv6, resolved via derived AAAA records? Hmm... 🤔

I am still not sure this is a good tradeoff...

netravnen, 2024-03-19:
Caveat,

How many local devices have Privacy Extensions enabled nowadays? 🤔 

Apple defaults to using privacy extensions on all platforms, afaik.

↪ DHCP(hostname)-to-DNS for subnets with SLAAC for clients to generate their addresses does not seem viable. If my client rotates it's primary v6 every few hours.