eworm-de/routeros-scripts #50: DHCP-to-DNS script does not differentiate same MAC addresses on different network domains

KuxaBeast, 2023-10-22:
Instead of properly differentiating between different DHCP network domains and assigning different records for different leases with same MAC addresses, the script ignores other addresses. This feature is particularly useful when using multiple VLANs on devices.

#49 is also affected by this.

eworm-de, 2023-10-27:
This behavior has historical reasons... Some time ago it was not possible to have different DNS name suffixes for different (DHCP) networks. Support for that was added later.

We should probably change that now... Though care has to be taken in case the DNS name suffix for different networks is still configured to be the same.

eworm-de, 2023-10-27:
Fixed with 3474b9a15c75ba4bb0af438362cd096be848da91.

closed 2023-10-27 by eworm-de

KuxaBeast, 2023-10-27:
@eworm-de Actually I made the fix as part of #49 already :D. I will check out your changes and perhaps rebase the MR upon it?

eworm-de, 2023-10-27:
Yes, but it was on top of the IPv6 changes and I preferred some details to have different. For example I decided to go with a comment the can be parsed by `$ParseKeyValueStore`.

Please test and report back (or prepare patches) if things are still misbehaving.