eworm-de/routeros-scripts #56: check-routeros-update and other similar scripts suggested scheduler time

bubenkoff, 2024-01-25:
In the check-routeros-update script's docs, it's suggested to use the 'startup' time of the script to be scheduled.
For such actions, it seems more logical to schedule it at night instead, as the update shouldn't normally happen in the busy time.

eworm-de, 2024-01-25:
The primary intend here is to notify the user on update. If device is configured to install updates automatically it may make sense to schedule at night. You are free to change that to your needs.

bubenkoff, 2024-01-25:
Still IMO changing the suggested time to the night would serve both
purposes just fine.

eworm-de, 2024-01-25:
Hmm, we have a mechanism to schedule the reboot at night... Perhaps we should just add a confirmation option to always schedule the reboot on automated update?

bubenkoff, 2024-01-25:
Do you mean to schedule it with this change, instead of directly rebooting?

bubenkoff, 2024-01-25:
> Yes. Packages are installed immediately, and reboot is scheduled if
> configured - just like you do when running packages-update interactively
> from terminal.
>
> There is another reason I want the check to be scheduled after startup: I
> want to avoid making all devices in a network to download the packages at
> the exact time. Scheduling that way is easy to achieve it.

Ok, thanks for the clarification.
I think it makes a lot of sense to implement it this way indeed.

eworm-de, 2024-01-26:
Pushed changes to the `next` branch... Do you want to give it a try?

bubenkoff, 2024-01-26:
Yes, please!

closed 2024-01-30 by eworm-de

bubenkoff, 2024-02-08:
@eworm-de the router got autoupdated last night, and the ```:global PackagesUpdateDeferReboot true;``` in the config didn't seem to defer the upgrade, it seemed to reboot right away when the script was scheduled (2 am).
Is it a good idea to add some logging so that we can see that the defer really happens?

eworm-de, 2024-02-08:
There is logging, it gives something like:
```
script,info packages-update: Scheduled reboot for update at 03:25:44 local time (Europe/Berlin).
```
And it works for me.

bubenkoff, 2024-02-08:
Ok thanks, maybe I've missed it indeed, as I only have a memory log there.
I think I'll add packages-update to the log forwarding to monitor it better.

commit 9d037ac9 (2024-01-26): packages-update: support deferred reboot on auto-update

Closes #56
commit 45c6fd33 (2024-01-26): packages-update: support deferred reboot on auto-update

Closes #56
commit 489a95c7 (2024-01-26): packages-update: support deferred reboot on auto-update

Closes #56
commit 1f15a38c (2024-01-29): packages-update: support deferred reboot on auto-update

Closes #56