eworm-de/routeros-scripts #59: ADD username/password auth for ntfy

gionag, 2024-04-01:
Hello !
can you provide a way to specify username and password for ntfy ?
fetch can accomodate user, pass for that.

some private ntfy instances (self-hosted) can be secured by that

thanks

eworm-de, 2024-04-01:
I do not have access to that for testing.

gionag, 2024-04-01:
i will test on my side and make a pull req

eworm-de, 2024-04-08:
This is just *basic auth*, no? Wondering if giving credentials with the domain works... Can you try something like "`username:password@domain`"?

eworm-de, 2024-04-13:
Ping... Any chance you test this?

Note to self... This is the documentation on authentication: https://docs.ntfy.sh/publish/#authentication

gionag, 2024-04-14:
tested.
works perfectly !
good job and nice addition.

(the more i use your scripts ecosystem more i like them, keep it up !)

eworm-de, 2024-04-14:
Great, thanks for testing!

commit ca2e5f2a (2024-04-15): mod/notification-ntfy: support basic auth

Closes #59
closed 2024-04-15 by eworm-de