eworm-de/routeros-scripts #60: Failed Telegram Notification

gionag, 2024-04-01:
Testing the telegram notification.
triple checked bot chatId and BotToken, they are correct.
postman works with that parameters.

notification is queued and never received.

i can see that it is downloading a GOdaddy cert, CRL is in red.

[image]
[image]

have you recently tested it ?

eworm-de, 2024-04-01:
I use it all the time, so absolutely sure the code works.

CRL is not used, as it causes too much trouble. All my scripts run `fetch` with `check-certificate=yes-without-crl`.

Does this give the expected output?

```
:put $TelegramTokenId;
:put $TelegramChatId;
```

eworm-de, 2024-04-02:
Anything in logs that gives a clue? Running `fetch` always leaves some messages there.

gionag, 2024-04-02:
> I use it all the time, so absolutely sure the code works.
> 
> CRL is not used, as it causes too much trouble. All my scripts run `fetch` with `check-certificate=yes-without-crl`.
> 
> Does this give the expected output?
> 
> ```
> :put $TelegramTokenId;
> :put $TelegramChatId;
> ```
```
:put $TelegramChatId;      
3#######8
:put $TelegramTokenId;     
###########:AAAAAAAAAAAAA_BBBBBBBBBBBBBBBBBBBBBBBB
```

in logs i have 

`fetch,info Download from api.telegram.org FAILED: Fetch failed with status 400`

if i ping api.telegram.org it pings, and resolve to the correct ip

The bot is not a member of any group. and the option i have set in "globa-config-overlay" are just
```
:global TelegramTokenId "TOKEN_ID_AS_ABOVE";
:global TelegramChatId "CHAT_ID_AS_ABOVE";
```

What else could i check ?

gionag, 2024-04-02:
> Just to be sure, as not visible from your output... Your token does not start with "bot", no? That belongs to the url and is added automatically.

indeed is not.

eworm-de, 2024-04-02:
Running out of ideas... Please double-check there's no typo, missing or extra character. It's no more than these two variables.

eworm-de, 2024-04-08:
Any news on this? Did you figure?

gionag, 2024-04-08:
it solved itself. 
So i haven't figured it out.
nothing was changed on my side... maybe an auto update of the script has changed some sub-section that was key to make it work.

eworm-de, 2024-04-08:
Hmm, there were no changes related to Telegram notifications... No idea what happened then.

Anyway, closing the issue.

closed 2024-04-08 by eworm-de