eworm-de/routeros-scripts #62: notification doesn't stat from netwatch

gionag, 2024-04-27:
i have a very simple monitor via netwatch

preface:
i have notification enabled via mail, ntfy and telegram.
via terminal, if i do a simple
`$SendNotification "test" "test"`
everything works !

on a Netwatch monitor i have set for UP: 
```
:global SendNotification
$SendNotification "Test" "Up";
```

for DOWN:
```
:global SendNotification
$SendNotification "Test" "Down";
```

but on netwatch it doesn't work.
tested by logging something, and it works fine.
any suggestions on why it happens ?

eworm-de, 2024-04-27:
Probably a permission problem... Using `fetch` requires elevated permission, I think `test` or something. This is not granted for scripts in netwatch.

Just use my script for this: https://git.eworm.de/cgit/routeros-scripts/about/doc/netwatch-notify.md

gionag, 2024-04-27:
i have previously used this kind of code in netwatch
```
:local BotToken "BOT_TOKEN";
:local ChatID "CHAT_ID";
:local ParseMode "html";
:local DisableWebPagePreview True;
:local SendText $MessageText;

:local tgUrl "https://api.telegram.org/bot$BotToken/sendMessage?chat_id=$ChatID&text=$SendText&parse_mode=$ParseMode&disable_web_page_preview=$DisableWebPagePreview";

/tool fetch http-method=get url=$tgUrl keep-result=no;

:log info "Send Telegram Message: $MessageText";
```
and it worked fine...

eworm-de, 2024-04-28:
From [Mikrotik netwatch documentation](https://help.mikrotik.com/docs/display/ROS/Netwatch):

> Netwatch is limited to read,write,test,reboot script policies. If the owner of the script does not have enough permissions to execute a certain command in the script, then the script will not be executed. If the script has greater policies than read,write,test,reboot - then the script will not be executed as well, make sure your scripts do not exceed the mentioned policies.

If this worked for you... Probably with an older RouterOS version before that limitation.

You should really check out my script linked above.

closed 2024-04-28 by eworm-de