eworm-de/routeros-scripts #63: Feature request - optional backup-upload notification

pniewiadowski, 2024-05-06:
Do you think you can add an option to enable/disable notification for backup-upload? This is something I would like to disable as I don't want to receive notification if backup upload every single time. What do you think?

eworm-de, 2024-05-06:
I do not want to add on option like that. IMHO it is important to receive notifications to make sure you actually do have a backup. Nothing is worse than requiring a backup, and expecting to to have one - but in reality it failed since ages.

Even not sending a notification on successful can be fatal. Either everything works, or both fails - the backup and the notification. Notification on successful backup are sent silent where supported, though - so the "pling" is omitted for Telegram notifications for example.

However you can still configure what you want, if you really want. Have a look at `global-config` and the hints on `Override` variables:

* Send notifications for backup scripts to another address/account/group and discard them there.
* Do not configure notification settings globally, but only for scripts you want notifications for.

closed 2024-05-06 by eworm-de

pniewiadowski, 2024-05-06:
Imagine getting notification every day, that your backup finished successfully. At some point you start to ignore the messages in some way. Or you don't even notice the notification should arrive but it doesn't. I am the guy who want to receive notification if something failed and at the same time, have another check in place if the backup really works. But this is my point of view.

Thanks for all the scripts. These are awesome.