eworm-de/routeros-scripts #64: backup-email / backup-upload

dd2594gh, 2024-05-07:
When attempting ton install/run these scripts on a new CRS309 I am unable to get them to run. The scripts save out to tmpfs -- which on a CRS317 there is when I look a tmpfs shows as a disk. With the fresh install on the new CRS309 I have no tmpfs. Where/when is this created? Have I missed something?

dd2594gh, 2024-05-07:
For information this is CRS309-1G-8S+ router/switch. It is a brand new switch with minimal (i.e., IP Address, etc. to get updates and internet access). Running RouterOS 7.14.3

Executed /system/script/run backup-email; -- let it run for over 10 minutes with no output to the terminal, no creation of tmpfs or files created. The log in winbox from the CRS309 shows this 8 times over the 10 minutes (appears to execute twice every 3 minutes):

ntp settings changed by winbox-3.40/tcp-msg(winbox):admin@aaa.bbb.ccc.ddd/terminal/script:backup-email (/system ntp client set enabled=no; /system ntp server set)

Executed /system/script/run backup-upload; -- let it run for over 10 minutes with no output to the terminal, no creation of tmpfs or files created. The log in winbox from the CRS309 shows this 8 times over the 10 minutes (appears to execute twice every 3 minutes):

ntp settings changed by winbox-3.40/tcp-msg(winbox):admin@aaa.bbb.ccc.ddd/terminal/script:backup-upload (/system ntp client set enabled=no; /system ntp server set)

Executed /system/script/run check-routeros-update; -- let it run for over 10 minutes with no output to the terminal. The log in winbox from the CRS309 shows this 8 times over the 10 minutes (appears to execute twice every 3 minutes):

ntp settings changed by winbox-3.40/tcp-msg(winbox):admin@aaa.bbb.ccc.ddd/terminal/script:check-routeros-update (/system ntp client set enabled=no; /system ntp server set)

For comparison on a working system (CRS317):

Executing /system/script/run backup-email; -- produced nothing to the terminal but executed within 10 seconds. The following entries were produced in the log:

new script scheduled by winbox-3.40/tcp-msg(winbox):admin@192.168.111.190/terminal/script:backup-email (*18 = /system scheduler add comment="Queuing new mail..." interval=1s name=_FlushEmailQueue on-event=":global FlushEmailQueue; \$FlushEmailQueue;" start-time=startup)

changed scheduled script settings by scheduler:_FlushEmailQueue/action:45577 (/system scheduler set _FlushEmailQueue comment="Doing initial checks..." interval=1m)

changed scheduled script settings by scheduler:_FlushEmailQueue/action:45578 (/system scheduler set _FlushEmailQueue comment=Sending... interval=1m)

sent <=?utf-8?Q?[sw0] =F0=9F=92=BE=EF=B8=8F=F0=9F=93=A8=EF=B8=8F Backup & Config?=> to: devin.dean@live.com 

script removed from scheduler by scheduler:_FlushEmailQueue/action:45581 (/system scheduler remove *18)

Executed /system/script/run backup-upload; This script produced 2 groupings of status, uploaded, total, duration. Executed in under 10 seconds and produced much the same log entries as the email version (with some changes and specifics for upload).

Executed /system/script/run check-routeros-update; -- script produced a single line of output to the terminal: info: System is already up to date. It also produced a single log entry: check-routeros-update: System is already up to date.

The CRS317-1G-16S+ is running RouterOS 7.14.3. 

Do I need to remove all of the scripts from the CRS309 and try installing again?

eworm-de, 2024-05-07:
Can you please give the output of:
```
:put [ $IsTimeSync ];
```

dd2594gh, 2024-05-07:
Output is "false"

eworm-de, 2024-05-07:
I guess your device has a ntp (time) server configured, but fails to sync the time. Thus the scripts wait for the time to be synced. Please check your ntp settings in `/system/ntp/client/`...

dd2594gh, 2024-05-07:
Thanks - I will look into that ... strange it has NTP servers and is enabled, but status is waiting

closed 2024-05-07 by eworm-de