eworm-de/routeros-scripts #67: Self-hosted ntfy.sh certificate issue

tryallthethings, 2024-06-01:
Hey! 
I just found this and wanted to set up ntfy.sh notifications. But ran into an issue. I have a self-hosted ntfy.sh instance. I downloaded the main script, added the ntfy.sh module, configured it and tried it. But it runs into an error.

```
# Overlay for global configuration by RouterOS Scripts
# Copyright (c) 2013-2024 Christian Hesse <mail@eworm.de>
# https://git.eworm.de/cgit/routeros-scripts/about/COPYING.md
#
# global configuration, custom overlay
# https://git.eworm.de/cgit/routeros-scripts/about/#editing-configuration

# Copy relevant configuration from global-config, paste and modify it here.
# https://git.eworm.de/cgit/routeros-scripts/about/global-config.rsc

:global NtfyServer "n.domain.com";
:global NtfyServerUser [];
:global NtfyServerPass [];
:global NtfyTopic "Mikrotik";

# End of global-config-overlay
```

Error: 

`Download from n.domain.com FAILED: SSL: ssl: no trusted CA certificate found (6)`

Do I have to manually download and update the certificate + root certificate to make this work, or is there an automatism? I'm using Let's encrypt for the service.

eworm-de, 2024-06-01:
There is no automatism in this cast, as the script does not know what certificate is required (and I do have a small subset in my repository only).

So in general you would have to do it yourself manually, however for _Let's Encrypt_ you can just run:
```
$CertificateAvailable "R3";
```

tryallthethings, 2024-06-01:
Might be a good addition for the future to have at least the big CA's automated.

The command worked great. Can I throw this into the scheduler to have it updated regularly?

eworm-de, 2024-06-01:
No need... The certificate is valid for loooooong time.

And you could switch to ECDSA certificate chain (root X2) to be on the safe side. That's what I use for my site.

closed 2024-06-01 by eworm-de