eworm-de/routeros-scripts #68: Restore LogPrintExit2 for backward compatibility

Kentzo, 2024-06-04:
The changelog says:

> info: Change 123: Introduced new function '$LogPrint', and deprecated '$LogPrintExit2'. Please update custom scripts if you use it.

Please correct me if I'm wrong, but this function was not deprecated but outright removed. Please restore it for compatibility, I have a bunch of scripts that depend on it.

eworm-de, 2024-06-04:
Well, that's not true. The function was deprecated in 1feeed145d6b63ef60fc3c0abbe0703b7bd73837, and a warning was added in 576a2c97cf88b7d1b36907469fb6fd34618523b3. The final removal happened in 36cf4d028bcf9597143249aed238dc5e131fda50, about a month later.

But you are right that all this happened in a quite small time frame. Usually I keep deprecated function around for longer, this time I wanted to get rid of it as soon as possible as it can cause some trouble with RouterOS 7.15.

Feel free to restore the function on your devices. To make `global-functions` auto-load it create a script `mod/LogPrintExit2` with this content:
```
# dropped upstream, but restored
:global LogPrintExit2 do={
  :local Severity [ :tostr $1 ];
  :local Name     [ :tostr $2 ];
  :local Message  [ :tostr $3 ];
  :local Exit     [ :tostr $4 ];

  :global LogPrint;

  $LogPrint $Severity $Name $Message;

  :if ($Exit = "true") do={
    :error ("Hard error to exit.");
  }
}
```

closed 2024-06-04 by eworm-de

Kentzo, 2024-06-04:
What was the issue you were hitting in RouterOS 7.15? I may need to revisit my scripts too.

eworm-de, 2024-06-04:
The problem with RouterOS scripting is that it lacks some control structures, including something like `exit` to leave a script early without error, `elif` to complete `if` ... `else`, `continue` or `next` and `break` or `last` for loops, ...

For a long time I used `:error` to exit a script early, as nothing did care about the returned error. The function `LogPrintExit2` did make use of that as well. Now with RouterOS 7.15 this results in a lot of...
```
Executing script from scheduler failed, please check it manually
```
Obviously I do not want my users to be flooded with that message, and I updated all my scripts. But I also wanted to make sure nobody suffers this when using a deprecated function.

That said... You can just replace the function call to `LogPrintExit2` with `LogPrint`, except that the last parameter is ignored and exit functionality is gone. If you did use that you need to find another way to exit your script.