eworm-de/routeros-scripts #74: version 7.15.2 - crash and error

Simon-CR, 2024-07-08:
if I use terminal command to automatically download the scripts, all 3 get installed, however I cannot modify the global-functions (crashed my winbox with script error: error - contact MikroTik support and send a supout file (10)

if I get the content from the script from GitHub and paste it in a new script and try to run it, I get the syntax error (line 1046 column 90) error when trying to run it.

thanks

Simon-CR, 2024-07-08:
so somehow after running the hello-world test ($ScriptInstallUpdate hello-world "base-url=https://git.eworm.de/cgit/routeros-scripts-custom/plain/, url-suffix=?h=main") it replaced the global-functions script again ..  I can't edit the file without winbox crashing but I seem to be able to use what's in it ..

eworm-de, 2024-07-08:
Well, the script containing global functions is quite huge, and I do not recommend editing it on the device - neither with Winbox, nor on terminal. Some devices will probably just run out of memory, some may not. Even then Mikrotik might have limited the resources for certain actions, to mitigate resource exhaustion.

I do edit the scripts (and especially the global functions) on my workstation only. Scripts are managed in a git repository, and device download them from there, even for testing.

So there is not a lot I can do here. If you are concerned about the crash report it to Mikrotik...

closed 2024-07-08 by eworm-de