eworm-de/routeros-scripts #76: RoutesOS 6 Branch

dmibot, 2024-08-10:
```
foreach Script in={ "global-config"; "global-config-overlay"; 
"global-functions" } do={ / system script add name=$Script source=([ / tool fetch 
check-certificate=yes-without-crl ("https://git.eworm.de/cgit/routeros-scripts/pla
in/" . $Script . "\?h=routeros-v6") output=user as-value]->"data"); };

```
failure: ssl connection error: handshake failed: unable to get local issuer certif
icate (6

eworm-de, 2024-08-10:
Oh, perhaps I should fix that at least...

Well, you need the new root certificate, "ISRG Root X2". It should work if you import that.

closed 2024-08-10 by eworm-de

eworm-de, 2024-08-19:
I updated the certificate stuff in 2cda1e9c4c3dd16c72c457e7ad2061da4992941f... Should just work now again.