eworm-de/routeros-scripts #77: Backup Upload via SFTP not working

rpchan44, 2024-08-20:
`:global BackupRandomDelay "30";
:global BackupUploadUrl "sftp://192.168.99.4/";
:global BackupUploadUser "z1";
:global BackupUploadPass "p@ssw0rd23";`

Using normal SFTP i was able to upload file using "z1" as user

This is the error has been log in MT during backup script execution

Upload to 192.168.99.4 FAILED: closing connection: <permission denied> 192.168.99.4:22 (6) (6)

eworm-de, 2024-08-20:
Hmm, not absolutely sure, but "`permission denied`" sounds like file permission... So I guess the authentication succeeded, but file could not be written. Possibly you missed the remote path or the destination folder is not writable? Any logs from the server that indicate what the issue could be?

rpchan44, 2024-08-20:
Thanks for your prompt reply, I was confused too
[image]

eworm-de, 2024-08-20:
Not sure if implementations differ in handling of default directory. You could try:
```
:global BackupUploadUrl "sftp://192.168.99.4/home/z1/";
```
Does that work?

rpchan44, 2024-08-20:
Yes that fixed it!!!! the sftp client implementation from mikrotik is flawed? anyway thanks a lot for your great effort!! please close this now

eworm-de, 2024-08-20:
No, it's ok - it just expects the complete path.

Have a look at your screenshots... For `scp` your added `~` which is equivalent to `/home/z1/` here. The interactive `sftp` just opens at the home directory automatically.

closed 2024-08-20 by eworm-de