eworm-de/routeros-scripts #79: [Feature Request] fw-addr-lists : JSON parsing

BadFlo, 2024-08-30:
Hi,

spamhaus.org offers IPv6 blocking list in JSON format only.

I don't think it would require a lot of work to add the functionality.

I'll get on it and try to create a Pull Request.

eworm-de, 2024-09-05:
Oh well. 😳😕
Actually this is not one JSON document, but one per line. Why do they do that!? Makes it a lot harder to sparse, as we can not deserialize with RouterOS's functionality in one go.

Also... Is this format specified anywhere or is it a proprietary invention by them?

eworm-de, 2024-09-05:
Can you give ef2d695c23a8df1efa6096c323583641a8c8182a a try? (The config should look like that in 83dceee6fd89d42216e8edbe8599b02c71151abf.)

See [info on branches](https://github.com/eworm-de/routeros-scripts/blob/main/BRANCHES.md) for an easy way to get the code to your device.

BadFlo, 2024-09-05:
@eworm-de, looks perfect. Same result as with my code, but yours is cleaner and may even work better for other files.

But not for [ipblocklist_recommended.json from feodotracker.abuse.ch](https://feodotracker.abuse.ch/downloads/ipblocklist_recommended.json)

Right now it's not a great testfile as there is only one entry ...I'll monitor and adapt your code to work with this (better) format too.

Regarding your question about the json format, I didn't find an answer on their website. I'll ask them by mail and keep you posted if I find anything interessting.

closed 2024-09-11 by eworm-de