eworm-de/routeros-scripts #80: Telegram notifications failure

sergiobaiao, 2024-08-30:
Hi there,

After working flawless for several weeks, my telegram notification script is failing now with this:

Download from api.telegram.org FAILED: 401 should contain www-authenticate header

what should i do?

eworm-de, 2024-08-30:
Hmm, very interesting... I have not seen something like that with Telegram API and I can not reproduce.

Just wondering is there a proxy between your Mikrotik device and the Telegram servers that requires authentication?

eworm-de, 2024-08-30:
Oh, and please check that variables (`TelegramTokenId` and `TelegramChatId`) are properly set and populated. An internal process crashing could clear these.

eworm-de, 2024-09-05:
No answer, so closing for now.

closed 2024-09-05 by eworm-de