eworm-de/routeros-scripts #85: ipv6-update issue

mikeS7, 2024-10-22:
My ISP doesn’t change ipv6 prefix on reboot and I’ve got an issue:

I receive the same /56 global prefix but prefixes for interfaces will change in case of different initializations order. Could you please force update on first run?

Another small issue, looks like DHCP Clients calls the script twice if I request both address and prefix  I’ve got an error "This script is supposed to run from ipv6 dhcp-client." in such case.

eworm-de, 2024-10-22:
> My ISP doesn’t change ipv6 prefix on reboot and I’ve got an issue:
> 
> I receive the same /56 global prefix but prefixes for interfaces will change in case of different initializations order. Could you please force update on first run?

Hmm, will have to think about this one... 🤔

The problem is that the script does not know if it is run first time... Though you could always remove the address-list entry. Something like this, where $Pool is your pool name:

```
/ipv6/firewall/address-list/remove [ find where comment=("ipv6-pool-" . $Pool) ];
```

> Another small issue, looks like DHCP Clients calls the script twice if I request both address and prefix I’ve got an error "This script is supposed to run from ipv6 dhcp-client." in such case.

Should be fixed with 06cbb7dc64fbca9174835acb2c9d865a597194c8... It just gives an info and exits. Can you check?

Though wondering... For me RouterOS does allow just one dhcp client for each interface. How did you configure this?

eworm-de, 2024-10-22:
Hmm, I hope that 6e85797d07677a533215ea770eac04f1b0474554 fixes your primary issue. Again, can you verify?

Note that you have to remove the existing static address-list entry for this to work.

mikeS7, 2024-10-23:
> Should be fixed with [06cbb7d](https://github.com/eworm-de/routeros-scripts/commit/06cbb7dc64fbca9174835acb2c9d865a597194c8)... It just gives an info and exits. Can you check?
> 

I doesn't ;)
Type of na-address is 'str'.

May I suggest to use the code:
`  :if (([ :typeof $PdPrefix ] = "nothing") && ([ :typeof $NaAddress ] = "nothing")) do={`
`    $LogPrint error $ScriptName ("This script is supposed to run from ipv6 dhcp-client.");`
`    :error false;`
`  }`

> Though wondering... For me RouterOS does allow just one dhcp client for each interface. How did you configure this?
>
I have only one in interface but request both address and prefix.
DHCP script is called multiple times on release. I count 4 calls:
na-valid - 0
pd-valid - 0
na-valid - 1
pd-valid - 1

On renew just one call.

I have no idea is it required to process `pd-valid = 0` case.

mikeS7, 2024-10-23:
> Hmm, I hope that [6e85797](https://github.com/eworm-de/routeros-scripts/commit/6e85797d07677a533215ea770eac04f1b0474554) fixes your primary issue. Again, can you verify?
> 
> Note that you have to remove the existing static address-list entry for this to work.

Verified, working without issue.
But docs a bit confused for me, it is not clear that static pool should be deleted with update.

Thank you very much for your scripts, its saved a lot of time

eworm-de, 2024-10-23:
> I doesn't ;) Type of na-address is 'str'.
>
> May I suggest to use the code: ` :if (([ :typeof $PdPrefix ] = "nothing") && ([ :typeof $NaAddress ] = "nothing")) do={` ` $LogPrint error $ScriptName [...]

I want do give different output on acquired address, and exit early.

This should work with d022c876514107bd345a17a0c01896947a83b45f now.

eworm-de, 2024-10-23:
> Verified, working without issue. But docs a bit confused for me, it is not clear that static pool should be deleted with update.

Updated the documentation in 3fc45a0bdbd94df01c8193a376f8f2d993502222. Is it clearer now?

> Thank you very much for your scripts, its saved a lot of time

You are welcome. 😉

mikeS7, 2024-10-24:
> > > Should be fixed with [06cbb7d](https://github.com/eworm-de/routeros-scripts/commit/06cbb7dc64fbca9174835acb2c9d865a597194c8)... It just gives an info and exits. Can you check?
> > 
> > 
> > I doesn't ;) Type of na-address is 'str'.
> > May I suggest to use the code: ` :if (([ :typeof $PdPrefix ] = "nothing") && ([ :typeof $NaAddress ] = "nothing")) do={` ` $LogPrint error $ScriptName ("This script is supposed to run from ipv6 dhcp-client.");` ` :error false;` ` }`
> 
> I want do give different output on acquired address, and exit early.
> 
> This should work with [d022c87](https://github.com/eworm-de/routeros-scripts/commit/d022c876514107bd345a17a0c01896947a83b45f) now.

Yes, it works! Thank you.

closed 2024-10-24 by mikeS7

eworm-de, 2024-10-25:
The typo is fixed in latest commit. All this ha been merged into `main`, so is available with regular update.

Thanks for testing!