eworm-de/routeros-scripts #86: Add support for ntfy access tokens

ementat, 2024-12-13:
Currently mod/notification-ntfy supports only Basic Auth method through username and password, but ntfy documentation points to another auth method: access tokens.

Access tokens are useful to avoid having to configure your password across multiple publishing/subscribing applications. For instance, you may want to use a dedicated token to publish from your backup host, and one from your home automation system.

Once an access token is created, you can use it to authenticate against the ntfy server, e.g. when you publish or subscribe to topics. Access tokens are using [Bearer auth](https://swagger.io/docs/specification/authentication/bearer-authentication/). Below is an example of using access token:

`curl -H "Authorization: Bearer tk_AgQdq7mVBoFD37zQVN29RhuMzNIz2" -d "Look ma, with auth" https://ntfy.example.com/mysecrets`

And for POST:
```
POST /mysecrets HTTP/1.1
Host: ntfy.example.com
Authorization: Bearer tk_AgQdq7mVBoFD37zQVN29RhuMzNIz2

Look ma, with auth
```

Please add support for Access tokens to mod/notification-ntfy.

eworm-de, 2024-12-13:
Support for authentication (with username and password) was added in #59 ... Will have to take a look at the bearer auth thing.

eworm-de, 2024-12-13:
Is the token something you just have or does it need to be calculated? A username is not required when using the token, no?

eworm-de, 2024-12-13:
Hmm, in documentation I added:

> Even authentication via access token is possible, adding it as password with a blank username.

That does not work?

eworm-de, 2024-12-13:
Please have a look at [using branches](https://github.com/eworm-de/routeros-scripts/blob/main/BRANCHES.md), then switch to `next` branch and give it a try.

ementat, 2024-12-15:
> Please have a look at [using branches](https://github.com/eworm-de/routeros-scripts/blob/main/BRANCHES.md), then switch to `next` branch and give it a try.

I'm getting "warning: Syntax validation for script 'mod/notification-ntfy' failed! Ignoring!" error when trying to use next branch.

eworm-de, 2024-12-15:
Oh, shit... I should have tried to install at least.

I will fix it later today, stay tuned!

eworm-de, 2024-12-15:
Fixed, please try again.

ementat, 2024-12-17:
> Fixed, please try again.

Script downloading is working now. And I performed several tests with Access Token auth - also working great!

eworm-de, 2024-12-17:
Great, thanks for testing!
(And do not forget to star 🌟 the repository if you have not already!)

closed 2024-12-18 by eworm-de