eworm-de/routeros-scripts #90: backup-upload - error ... is not a file.

BadFlo, 2025-02-13:
Since v133, I have an error on backup-upload :

`Saving system configuration
Configuration backup saved
    status: finished
  uploaded: 495KiB
     total: 495KiB
  duration: 2s
error: File 'tmpfs/backup-upload/MikroTik-estraviz-ch.backup' is not a file.
    status: finished
  uploaded: 215KiB
     total: 215KiB
  duration: 1s
error: File 'tmpfs/backup-upload/MikroTik-estraviz-ch.rsc' is not a file.
    status: finished
  uploaded: 3KiB
     total: 3KiB
  duration: 1s
error: File 'tmpfs/backup-upload/MikroTik-estraviz-ch.conf' is not a file.
`

What can I check on my side and what do you need from me to fix this ?

eworm-de, 2025-02-13:
Hmm... 🤨
Please show output of:
```
/file/print where name~"tmpfs/backup-upload/";
```

eworm-de, 2025-02-13:
Ah, and what RouterOS version is that? Running latest beta release?

BadFlo, 2025-02-13:
@eworm-de,

No output from his command :
`/file/print where name~"tmpfs/backup-upload/";`

RouterOS version is 7.17.2

eworm-de, 2025-02-13:
Please enable debug output:
https://rsc.eworm.de/DEBUG.md

... and try again.

BadFlo, 2025-02-13:
@eworm-de,
Not much better ...

```
debug: Script backup-upload started from terminal.
debug: Making directory: tmpfs/backup-upload
Saving system configuration
Configuration backup saved
    status: finished
  uploaded: 495KiB
     total: 495KiB
  duration: 1s
debug: Removing file: tmpfs/backup-upload/MikroTik-local.backup
error: File 'tmpfs/backup-upload/MikroTik-local.backup' is not a file.
    status: finished
  uploaded: 215KiB
     total: 215KiB
  duration: 2s
debug: Removing file: tmpfs/backup-upload/MikroTik-local.rsc
error: File 'tmpfs/backup-upload/MikroTik-local.rsc' is not a file.
    status: finished
  uploaded: 3KiB
     total: 3KiB
  duration: 1s
debug: Removing file: tmpfs/backup-upload/MikroTik-local.conf
error: File 'tmpfs/backup-upload/MikroTik-local.conf' is not a file.
```

eworm-de, 2025-02-13:
Ah, I have an idea what's going on... Currently on the go, will fix it later.

Sadly my own server is out of order, so I did not catch that one.

BadFlo, 2025-02-13:
Just checked under "files" and the backup files are still being deleted properly.

[image]

eworm-de, 2025-02-13:
Yes, because removing the directory works, and that is recursively.

But I remove files one by one to minimize memory pressure on devices with little RAM.

eworm-de, 2025-02-13:
Fixed... Please update and you are good to go.

Thanks a lot for reporting!

BadFlo, 2025-02-14:
All good !
Thanks for the fix 👍🏼