eworm-de/routeros-scripts #93: packages-update error

bubenkoff, 2025-03-15:
I have had this for several days, so the automatic package update is failing consistently.

```
The log on Main contains these 3 messages after 1w2d22:25:16 uptime.

🟠️ 2025-03-15 02:00:43 script;warning $DownloadPackage: Downloading package file 'routeros' failed.
🔴️ 2025-03-15 02:00:43 script;error packages-update: Download for package routeros failed, update aborted.
ℹ️ 2025-03-15 02:00:43 script;info check-routeros-update: Waiting for system to reboot.
```

Installed:  7.18.1
Available:  7.18.2

Is there any other log to check?

eworm-de, 2025-03-15:
Hmm, worked for me on all devices. The fetch command itself should log to system log. Any clues there?

bubenkoff, 2025-03-15:
Unfortunately no, due to lots of 

```
Download from api.telegram.org FINISHED

```

from telegram-chat script
How do I get rid of those?

bubenkoff, 2025-03-15:
Ah ok, found it.
It's disk space issue, due to lots of leftover files in the firmware folder, are those due to the package update?

eworm-de, 2025-03-15:
Do you use `capsman-download-packages`? I think that was not named explicitly here. Already fixed with ce39b79f69f978a8de521643b07b9c19de02b06f then.
Update and clean up, and you are good to go.

bubenkoff, 2025-03-15:
Yes, I use it. Thank you!