eworm-de/routeros-scripts #95: WaitMax should be $2 not $3

Kentzo, 2025-04-02:
https://github.com/eworm-de/routeros-scripts/blame/b560ea4b7d3da75b5f54633eea6a5fb7472d5bd6/global-functions.rsc#L1391

eworm-de, 2025-04-02:
Ah, right... Good catch! Do you want to create a pull request?

eworm-de, 2025-04-02:
... or alternatively - do you want me to make you the author of the commit?

Kentzo, 2025-04-02:
I can do a PR, np.

Before I do that, I'd like to add support for an additional argument: duration of the delay. My scripts usually take constant time to execute and it seems like a waste to poll every 100ms for something that is known to take 5s. Does it sound good to you?

eworm-de, 2025-04-03:
Hmm, not sure about that... Perhaps you should just call `:delay 5s` before if you know what time it takes.

Also the polling time is expended if you give more time, no? Perhaps you should give a longer time if concerned about polling.

eworm-de, 2025-04-08:
Ah, I fooled myself... This function always used a fixed delay. Please have a look at the changes in `next`, it has an increasing delay (longer polling interval) with longer wait time. Is that fine for you?

Kentzo, 2025-04-10:
This works 👍

closed 2025-04-14 by eworm-de