eworm-de/routeros-scripts #101: Send cmd to group another user not turst

taehyongsuzuki, 2025-05-11:
Hello, which config should be change when allow all member on group send the command.

eworm-de, 2025-05-12:
Kind of missing the context... Is this question about `telegram-chat`?

taehyongsuzuki, 2025-05-12:
oh i'm sorry.
yes correct telegram-chat

eworm-de, 2025-05-12:
Well, first of all just taking the sending user is technically the most straight forward.

Second it is more or less intended to add explicit trust there. After all you have full administrative control over the device(s)... Just adding another person to a group would give her/him full control as well.

If you want to manage that easily... The `global-config-overlay` can have snippets. Just add a script named `global-config-overlay.d/TelegramChatIdsTrusted`, with that content:
```
:global TelegramChatIdsTrusted { "user_A", "eworm_de", "user_B", ... };
```

You can then just update that script to add (or remove) trusted users (alias ChatIds). This even works via `telegram-chat`, with something like this:
```
/system/script/set source=":global TelegramChatIdsTrusted { \"user_A\", \"eworm_de\", \"user_B\", ... };" global-config-overlay.d/TelegramChatIdsTrusted;
/system/script/run global-config-overlay.d/TelegramChatIdsTrusted;
```

I guess trusting the members of a group would be possible... But wondering if there could be security related implications that I did not yet think of...

eworm-de, 2025-05-12:
You could try the change from b88c2a14cce830be0846315d1f8d0bc559cd8d12... That should work for what you want.

This is currently in `next` branch, not sure the change will make it into `main`.

taehyongsuzuki, 2025-05-12:
> :global TelegramChatIdsTrusted { "user_A"; "eworm_de"; "user_B"; ... };
I think this method is enough, and can group user access for each device.
because we manage 5-10 device on the group telegram as a joint project..

thanks...Christian Hesse 🙏

closed 2025-05-22 by eworm-de